esphome/esp-web-tools #375: Upload fails with ESP32-S2 native USB

jonasniesner, 2023-05-18:
When trying to use esp-web.tools to upload to an esp32-s2 with its onboard usb, the upload fails.

On Windows (Chrome or Edge), I get the following error

ESP Web Tools 9.2.1 by Nabu Casa; https://esphome.github.io/esp-web-tools/
install-dialog-2aee1312.js:424 esptool.js
install-dialog-2aee1312.js:424 Serial port WebSerial VendorID 0x303a ProductID 0x2
install-dialog-2aee1312.js:424 Connecting...
install-dialog-2aee1312.js:424 .
install-dialog-2aee1312.js:424 Detecting chip type... 
install-dialog-2aee1312.js:424 ESP32-S2
install-dialog-2aee1312.js:424 Chip is ESP32-S2FH32
install-dialog-2aee1312.js:424 Features: Wi-Fi,Embedded 4MB Flash
install-dialog-2aee1312.js:424 Crystal is 40MHz
install-dialog-2aee1312.js:424 MAC: 48:27:e2:57:7c:10
install-dialog-2aee1312.js:424 Uploading stub...
install-dialog-2aee1312.js:724 Error: Timeout
    at ia.read (install-dialog-2aee1312.js:424:72641)
    at async ta.command (install-dialog-2aee1312.js:424:55233)
    at async ta.check_command (install-dialog-2aee1312.js:424:58833)
    at async ta.mem_block (install-dialog-2aee1312.js:424:59483)
    at async ta.run_stub (install-dialog-2aee1312.js:424:65415)
    at async ta.main_fn (install-dialog-2aee1312.js:424:66917)
    at async install-dialog-2aee1312.js:724:2934
(anonymous) @ install-dialog-2aee1312.js:724
await in (anonymous) (async)
_confirmInstall @ install-dialog-2aee1312.js:724
await in _confirmInstall (async)
handleEvent @ styles-e96248a3.js:1
install.openepaperlink.de/:1 Uncaught (in promise) DOMException: Failed to execute 'open' on 'SerialPort': Failed to open serial port.

And in 1 out of 10 times, the upload works, but only if i hold the GPIO 0 button. Uploading to ESP32 or Esp32-S3 works with no problems.

On Linux, I get this error:

ESP Web Tools 9.2.1 by Nabu Casa; https://esphome.github.io/esp-web-tools/
install-dialog-2aee1312.js:424 esptool.js
install-dialog-2aee1312.js:424 Serial port WebSerial VendorID 0x303a ProductID 0x2
install-dialog-2aee1312.js:424 Connecting...
install-dialog-2aee1312.js:424 .
install-dialog-2aee1312.js:424 .
install-dialog-2aee1312.js:424 .
install-dialog-2aee1312.js:424 .
install-dialog-2aee1312.js:424 .
install-dialog-2aee1312.js:424 

install-dialog-2aee1312.js:424 Detecting chip type... 
install-dialog-2aee1312.js:424 ESP32-S2
install-dialog-2aee1312.js:424 Chip is ESP32-S2FH32
install-dialog-2aee1312.js:424 Features: Wi-Fi,Embedded 4MB Flash
install-dialog-2aee1312.js:424 Crystal is 40MHz
install-dialog-2aee1312.js:424 MAC: 48:27:e2:57:7c:10
install-dialog-2aee1312.js:424 Uploading stub...
install-dialog-2aee1312.js:724 Error: Timeout
    at ia.read (install-dialog-2aee1312.js:424:72641)
    at async ta.command (install-dialog-2aee1312.js:424:55233)
    at async ta.check_command (install-dialog-2aee1312.js:424:58833)
    at async ta.mem_block (install-dialog-2aee1312.js:424:59483)
    at async ta.run_stub (install-dialog-2aee1312.js:424:65415)
    at async ta.main_fn (install-dialog-2aee1312.js:424:66917)
    at async install-dialog-2aee1312.js:724:2934
(anonymous) @ install-dialog-2aee1312.js:724
await in (anonymous) (async)
_confirmInstall @ install-dialog-2aee1312.js:724
await in _confirmInstall (async)
handleEvent @ styles-e96248a3.js:1
install-dialog-2aee1312.js:424 esptool.js
install-dialog-2aee1312.js:424 Serial port WebSerial VendorID 0x303a ProductID 0x2
install-dialog-2aee1312.js:424 Connecting...
install-dialog-2aee1312.js:724 DOMException: Failed to execute 'setSignals' on 'SerialPort': Failed to set control signals.
(anonymous) @ install-dialog-2aee1312.js:724
await in (anonymous) (async)
_confirmInstall @ install-dialog-2aee1312.js:724
await in _confirmInstall (async)
handleEvent @ styles-e96248a3.js:1
install-dialog-2aee1312.js:724 Uncaught (in promise) DOMException: Failed to execute 'setSignals' on 'SerialPort': Failed to set control signals.
await (async)
_confirmInstall @ install-dialog-2aee1312.js:724
await in _confirmInstall (async)
handleEvent @ styles-e96248a3.js:1
install-dialog-2aee1312.js:724 Uncaught (in promise) DOMException: Failed to execute 'open' on 'SerialPort': The port is already open.
    at https://install.openepaperlink.de/js/install-dialog-2aee1312.js:724:5560

Is this an issue of esptool-js ?

ssieb, 2023-05-18:
https://github.com/esphome/esp-web-tools/issues/374#issuecomment-1550285777

sblantipodi, 2023-05-26:
as wrote in the issue linked by ssieb, using web tools 8.0.6 somewhat fixes the problem.
hard to say why newer esp web tools 9.2.1 works worse than the older 8.0.6

EmileSpecialProducts, 2025-04-07:
It looks like this is resolved from 10.1.0 release. I have tested it on my [UltraWiFiDuck](https://emilespecialproducts.github.io/UltraWiFiDuck/upload.html) 
This is a KeyStroke injection tool [github](https://github.com/EmileSpecialProducts/UltraWiFiDuck)
There is also a PIO [script](https://github.com/EmileSpecialProducts/UltraWiFiDuck/blob/master/Scripts/extra_script.py) that will generate the Bin files for the esp-web-tools

AEFeinstein, 2026-01-15:
This is not resolved for me using 10.1.1 and a custom ESP32-S2 board. It does work using 8.0.1 though.

sblantipodi, 2026-01-15:
@AEFeinstein same here

AEFeinstein, 2026-01-15:
One other data point, the latest esptool-js (0.5.6) [example page](https://espressif.github.io/esptool-js/) works fine with my ESP32-S2, but esp-web-tools 10.1.1 is still using esptool-js 0.5.3. That's since been updated to 0.5.6 in 728d0ddcb21a0b82fa44ed76afd1b8b4486674d8, so maybe the next release will have better performance.

AEFeinstein, 2026-01-15:
OK, I figured out containers and the build script and `main` with `esptool-js` 0.5.6 seems to have the same issue still