esphome/esp-web-tools #377: Can't get web installer to work with ESP32 S3

glowingkitty, 2023-05-24:
For another open source software project I tried to set up an ESP32 S3 web installer. While the flashing works fine now, I cannot get the firmware to load properly. If the firmware is flashed via plattformio, it works perfectly fine. But when flashing via the web installer, I get one of the two errors, depending on the manifest.json.

If I use the files platfomio is using according to their logs (-v):

```
Auto-detected: /dev/cu.usbmodem1101
"/opt/homebrew/Cellar/platformio/6.1.0/libexec/bin/python3.10" "/Users/.../.platformio/packages/tool-esptoolpy@1.30300.0/esptool.py" --chip esp32s3 --port "/dev/cu.usbmodem1101" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 8MB 0x0000 /Users/.../.platformio/packages/framework-arduinoespressif32@3.20004.0/tools/sdk/esp32s3/bin/bootloader_qio_80m.bin 0x8000 "/Users/../.pio/build/.../partitions.bin" 0xe000 /Users/.../.platformio/packages/framework-arduinoespressif32@3.20004.0/tools/partitions/boot_app0.bin 0x10000 .pio/build/.../firmware.bin
esptool.py v3.3
```

and the manifest.json:
```
{
  "name": "Software",
  "version": "0.1",
  "home_assistant_domain": "software",
  "new_install_skip_erase": false,
  "builds": [
    {
      "chipFamily": "ESP32-S3",
      "parts": [
        { "path": "../bootloader/bootloader_qio_80m.bin", "offset": 0 },
        { "path": "../firmware/partitions.bin", "offset":32768 },
        { "path": "../firmware/boot_app0.bin", "offset":57344 },
        { "path": "../firmware/firmware.bin", "offset": 65536 }
      ]
    }
  ]
}
```

the flashing works, but the serial output of the esp32 s3 gives me only an infinite loop of:
`invalid header: 0xffffffff`

If I merge the bin files via:
`python3 -m esptool --chip ESP32-S3 merge_bin -o mergedfirmware.bin.bin --flash_mode qio --flash_size 8MB 0x0000 bootloader_qio_80m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin`

and use the manifest.json:
```
{
  "name": "Software",
  "version": "0.1",
  "home_assistant_domain": "software",
  "new_install_skip_erase": false,
  "builds": [
    {
      "chipFamily": "ESP32-S3",
      "parts": [
        { "path": "../firmware/mergedfirmware.bin", "offset":0 }
      ]
    }
  ]
}
```
the flashing works, but the serial output of the esp32 s3 is stuck in a boot loop it seems:
```
Reconnecting to /dev/cu.usbmodem1101     Connected!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x43c
ets_loader.c 78 
```

But using the ESP Home web installer on https://esphome.github.io/esp-web-tools/ the install of your software works perfectly fine and the firmware runs. How did you fix this issue? what do you do differently?

glowingkitty, 2023-05-24:
I fixed the problem! For everyone else having the same issue:

use the bootloader_dio_80m (dio, not qio) and dio flashing mode to merge the bin files and then flash that merged firmware via the web installer. That worked finally.

glowingkitty, 2023-05-24:
Correction, still not completely fixed. Now I am able to get the firmware to run sometimes (but not consistent), but even then, the filesystem cannot write.
The ESP32 S3 Serial output gives me:
```
./components/esp_littlefs/src/littlefs/lfs.c:1790:debug: Bad block at 0x0
./components/esp_littlefs/src/littlefs/lfs.c:1796:warn: Superblock 0x0 has become unwritable
E (1430) esp_littlefs: Failed to format filesystem
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
Ada
```

and sometimes I get the error again:
```
Reconnecting to /dev/cu.usbmodem1101     Connected!
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x39c
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:QIO, clock div:2
load:0x3fce3808,len:0x39c
ets_loader.c 78 
```

beniroquai, 2023-05-25:
I'm having a similar problem with one of Seeed's XIAO modules. They describe some details here: https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/#uf2-bootloader Though, I think this is not related. Anyway, I would like to bump in this thread ;-)

Jason2866, 2023-05-26:
Your flash mode is wrong
```
python3 -m esptool --chip ESP32-S3 merge_bin -o mergedfirmware.bin.bin --flash_mode qio --flash_size 8MB 0x0000 bootloader_qio_80m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin
```
Initial flash mode can not be QIO!! The flash mode is switched by the 2nd stage bootloader to QIO
So correct is
```
python3 -m esptool --chip ESP32-S3 merge_bin -o mergedfirmware.bin.bin --flash_mode dio --flash_size 8MB 0x0000 bootloader_qio_80m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin
```

andy778, 2023-07-16:
I have another bug regarding [ESP32 S3 Mini ](https://www.wemos.cc/en/latest/s3/s3_mini.html) but that seems to atleast be problem with the size of ram that is 4MB and not 8MB.

```
E (250) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377474
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
```

ssieb, 2023-07-16:
@andy778 that's not a bug in the web tools.  You have to use the correct firmware file for your device.

andy778, 2023-07-17:
> @andy778 that's not a bug in the web tools. You have to use the correct firmware file for your device.

Ok. I need to read upon how this all [work](https://esphome.github.io/esp-web-tools/), I think it's addition of another ESP32 S3 target to the manifest that is needed.

balloob, 2023-08-13:
Going to close this as initial issue resolved. We don't want this to be a catch all issue for any issue with S3. If you have a specific issue with steps to reproduce, please open a new issue.

closed 2023-08-13 by balloob