esphome/esp-web-tools #418: [FEATURE] Pass directly a JSON Manifest Object instead of Filepath on webInstallButton

rainerfritz, 2023-10-09:
Hi!

In my application, I am getting the manifest as JSON object from my nodejs server. Currently I am using the esp-web-install-button, which is really awesome, thanks! Currently I could only find the possibility to pass the pathname of the manifest file. Is there a possibility to pass the object, avoiding a physical for every version and board we are supporting. 

THX
Rainer

EmileSpecialProducts, 2025-04-07:
You can have a look what was done on the [https://tasmota.github.io/install/](https://tasmota.github.io/install/)

balloob, 2026-01-20:
See the [README](https://esphome.github.io/esp-web-tools/) for "Generating a manifest dynamically & version management"