esphome/esp-web-tools #432: [Feature] multiple variants of the firmware

cr1tbit, 2023-11-10:
Hello,

For my case I need multiple configuration options:

* Firmware revisions
* Firmware only / LittleFS partition only / full image

For now I'm just adding multiple install buttons on the website so the user can choose what to flash, but it would be more elegant, if the button could use a manifest with an array of possible configurations, and then create a dropdown to choose the specific one.

Is this something this project would benefit from? And is this achievable in foreseeable future?

balloob, 2023-11-13:
That's not a feature we want to support, as every option requires descriptions that have different requirements per firmware. Instead, create an interface to guide the user to pick the right option, like you can see here https://esphome.io/projects/

closed 2023-11-13 by balloob

TD-er, 2023-11-13:
You mean a bit like how I organized it here: https://td-er.nl/ESPEasy/latest/ ?

cr1tbit, 2023-11-19:
@TD-er YES! Thanks for the reference.

If at some point I'll approach this, I'll probably make a button matrix, with revisions as rows and variants (fullimage, firmware-only) as columns.

How cool it would be, if it could be integrated with releases from github CI... now I'm daydreaming.