esphome/esp-web-tools #455: [Feature] Support for ESP32-H2 and ESP32-S6

torbeneims, 2023-12-26:
I would like to use ESP Web Tools for H2 and S6 family boards. I see that these families are not currently supported. Do you plan on adding support for these boards? What would I need to do for them to be supported?

TD-er, 2023-12-27:
Huh? Never heard of any ESP32-S6 yet.
Do you have a link about those?

torbeneims, 2023-12-27:
> Huh? Never heard of any ESP32-S6 yet. Do you have a link about those?

You're right, meant to say ESP32-C6. Fixed that above.

balloob, 2024-01-18:
Have you tried it? [Upstream esptool-js supports them](https://github.com/espressif/esptool-js/tree/main/src/targets) and so I don't see why it wouldn't work. What error do you get?

lboue, 2024-03-28:
Is this 10.0.1 version deployed online on https://web.esphome.io?

dattasaurabh82, 2024-12-15:
Is esp32-H2 supported. I have the issue: [issues/537](https://github.com/esphome/esp-web-tools/issues/537)

ximex, 2025-02-01:
@torbeneims @dattasaurabh82 could this issue get closed?

dattasaurabh82, 2025-02-01:
> [@torbeneims](https://github.com/torbeneims) [@dattasaurabh82](https://github.com/dattasaurabh82) could this issue get closed?

Yes please. Have been solved

closed 2025-04-28 by jesserockz