esphome/esp-web-tools #464: Improv Wi-Fi over Serial fails with ESP32 devices

bwims, 2024-01-23:
Hello,

I am having a real problem with using the web based utility to install ESPHome on ESP32 devices.

I can install on a Wemos D1 (8266) with no trouble, but when I try with an ESP32-S2 or ESP32-S3, the software installs, but then I get the message “An error occurred. Improv Wi-Fi Serial not detected.”

Can you confirm whether this is expected, and, if not, how to debug it?

Thanks!

B.

balloob, 2024-01-24:
Make sure you run the latest version of ESPHome as we have made some fixes for those models. Our voice assistant ready-made project works just fine with ESP32-S3 boxes https://esphome.io/projects/

balloob, 2024-01-25:
Try the dev version. We skipped a release and it might not have been out yet.

bwims, 2024-01-25:
Sadly no.  Running Version: 2024.1.0-dev I still the error “An error occurred. Improv Wi-Fi Serial not detected.”

Just to make sure, I am trying to use https://web.esphome.io/?dashboard_wizard "Prepare For First Use"

with an ESP32 S2 Mini WIFI Board Based ESP32-S2FN4R2 4MB FLASH 2MB PSRAM 

that looks like this:

[image]

I know there are workarounds, but if I can help debug it, I am willing.  If there are any log files I can send, let me know where they are.

Thanks!

B.

balloob, 2024-01-26:
I don't know about that board. In your initial report you mentioned that S3 doesn't work, but that we have working. 

Try this URL https://esphome.io/projects/?type=empty

bwims, 2024-01-26:
I did also mention S2

> I can install on a Wemos D1 (8266) with no trouble, but when I try with an **_ESP32-S2_** or ESP32-S3, the software installs, but then I get the message “An error occurred. Improv Wi-Fi Serial not detected.”

Since then I moved over to OpenHASP for the S3 device

However, I had tried the url referenced before.   It performs the install ok, but afterwards there is not even an attempt at wifi configuration.  The only choices offered are to reinstall or to look at the logs as shown in the screenshot below.

[image]

I can get the device working and adopted by creating an uploadable .bin fil but it is a clunky way to go.

(I actually used a tip I got from Reddit [here ](https://www.reddit.com/r/Esphome/comments/11a08za/can_esp32s2_minis_be_used_for_esphome/) to get the board specification right:

```
esp32:
  board: lolin_s2_mini
  variant: esp32s2
  framework:
    type: esp-idf
```

So I take it that the S2 is not supported yet by web tools?  If so, can I make a plea for inclusion? I am willing to beta test!

Thanks!

B.

bwims, 2024-01-31:
Hmm.... so is the S2 supported?  Or not?

sblantipodi, 2024-04-12:
I have the same problem on ESP32-S3 running the [Glow Worm Luciferin firmware](https://sblantipodi.github.io/glow_worm_luciferin/beta.html)

There is a weird thing...
If I click the install button, select the device, 
I get the error:
`Improv initialization failed. Error: Improv Wi-Fi Serial not detected`
and there is no option to configure WiFi:
[image]

but if then I click `Logs and console` and then `Back`
the option to connect to WiFi is always there

[image]

@balloob sorry if I quote you but 
what does the back button that the install button does not?

ldijkman, 2024-04-12:
> I have the same problem on ESP32-S3 running the [Glow Worm Luciferin firmware](
> but if then I click `Logs and console` and then `Back` the option to connect to WiFi is always there

same for me on ESP32
switch to logs and console back and forth 
several times until configure wifi apears
not on esp8266 version
only on esp32
thought it was my arduino coding not good

 https://ldijkman.github.io/async-esp-fs-webserver/

TD-er, 2024-04-14:
I've also noticed this behavior.
Today I just looked into this a bit more and it looks like some kind of timeout issue in combination with not resetting the ESP right after flashing.

For example on an ESP32-C3 (USB via HWCDC, not USB to serial chip) I clicked to open the console in the webflasher menu right after flashing and I didn't even get my console output until I sent a reset via the terminal window. (thus not pressing RST button)

This could indicate the board may still have been in flash mode?
So maybe the webflasher should try a few times more to get the IMPROV response and/or need to wait a bit longer for sending the reset signal?

sblantipodi, 2024-06-02:
@TD-er have you tried a workaround for this problem?
it is very hard to configure WiFi in this way since requires a lot of trials and errors and I don't know if all users may be able to do it without constantly reflashing the device.

TD-er, 2024-06-02:
It is still a bit of trial-and-error as you need to toggle log-view, click "back" and see if IMPROV details are shown.

You can try and see for yourself with my ESPEasy web flasher: https://td-er.nl/ESPEasy/latest/

sblantipodi, 2024-08-13:
I need to admit that the Tasmota version of the Web Installer works much better than this.
Sadly it's impossible to flash a lot of devices with the esphome version...

omar10wahab, 2025-01-13:
> I've also noticed this behavior. Today I just looked into this a bit more and it looks like some kind of timeout issue in combination with not resetting the ESP right after flashing.
>
> For example on [...]

I want to emphasize what youre saying cause thats what worked for me. Power cycling/resetting/pulling usb from the computer and plugging back in multiple times eventually got everything to work for me. Weird thing was i could flash the ESPhome firmware with little to no issue using whatever steps needed but the wifi is finnicky. I thought i had a bad ESP-01S and then i went and tried a ESP-32-C3 Super mini (all these from alibaba) and it gave me the same weird wifi over serial fail message. Resetting the device multiple times until i saw logs populating with no errors and then trying made it finally work for me. I then went back to my ESP-01s and tried a similar process of unplugging and plugging back in and it finally took the wifi and connected.

Another thing to is sometimes ungplugging and restarting after everything is set still makes things work weird. Theres something to these devices like you said related to powering off incorrectly or timing out weirdly but i encourage people to try several times before giving up. especially if your board is being recognized and flashing ESP home properly.

mjhags, 2025-01-20:
Just been banging my head on this one.  To get it to work for me I literally just went to the Logs view (from https://web.esphome.io/?dashboard_wizard) which showed a blank log screen, hit Reset Device and the Logs sprang into life.  After that the Wifi install worked fine.

scpotter, 2025-01-22:
Also got the error “improv wifi serial not detected” flashing Sonoff SV v1 (esp8266) on Web.ESPhome.

Similar symptoms: Logs showed as blank (reset did nothing), select configure wifi from menu repeats the error.

Resolution/workaround: Pulled the 3.3 power to board from TTL, plugged back in, selecting configure wifi from the menu was a success.

After that I was able to take control in ESPHome builder, edit and download.bin. Attempted to Install through Web.ESPhome, TTL was unable to connect, not sure if that’s expected. The TTL being used: https://a.co/d/2XODQYg

OTA from ESPHome builder worked fine.

xbmcnut, 2025-02-21:
I spent hours trying to get my generic ESP32-C3's working and finally from info on this forum, I hit the reset device and everything sparked into life.

[image]

dattasaurabh82, 2025-03-07:
I'm having the same issue with Seeed Studio Xiao esp32s3-sense board. 
It was working but since today, not at all. 
After reading [this article](https://www.espboards.dev/blog/setting-up-esphome-first-time/), seems like it's not solvable ? 😭

ajfriesen, 2025-03-20:
I have an es32-c3 and flash it via a cp2102n.

It is the esp32-c3-mini-1u-h4 in a custom board.

However I always: get this message in the browser for the improv serial:

```
Initializing Improv Serial
install-dialog-CQzvAv3p.js:823 Improv initialization failed. Error: Improv Wi-Fi Serial not detected
    at install-dialog-CQzvAv3p.js:491:1210
```

Flashing works fine when I hit the boot button on my board, but the wifi setup does not.

Any idea how I can debug this further?

I have added the this to my esphome.yaml:

```
improv_serial:
```

TheOrangeStuff, 2025-03-25:
@ajfriesen 
Newb here. I'm having the same issue. Can you elaborate on your solution?
You had to specify `uart0` ... where? That looks like a yaml?
Sporting a Huzzah Esp32 v2 feather, connected via USB, trying to flash from web.esphome.io

Thanks in advance!

ajfriesen, 2025-03-25:
> @ajfriesen 
> Newb here. I'm having the same issue. Can you elaborate on your solution?
> You had to specify `uart0` ... where? That looks like a yaml?
> Sporting a Huzzah Esp32 v2 feather, connected via USB, trying to flash from web.esphome.io
> 
> Thanks in advance!

I have built a custom board myself. 
And I was not using the esp32 to flash the board.

My board was additionally connected to a uart chip.
And in order to have the wifi improv to work, you have to setup the above mentioned optien. Because the logger and the wifi improv are using the same uart then.

If your board has a second uart chip you can try the same.
If not, I can not help.

the78mole, 2025-03-26:
I also have a strange problem with my custom board. I have `esp32_improv` an `improv_serial` in my yaml, but neither serial improv nor BLE improv work.

https://github.com/the78mole/ESPhome-KM271-WiFi/blob/main/buderus-km271_en.factory.yaml

I spent hours trying to get it working. If I just use the example/demo firmware from web tools (https://esphome.github.io/esp-web-tools/), everything works (serial and BLE). If I do a minimal config in ESPhome in my home assistant add the improv lines and delete the WiFi client lines (keeping the fallback hotspot) it also works. But as soon as it comes to km271-config, it doesn't. I also checked, which configuration has been applied in the GitHub actions (creating the release files), but also there, the improv lines appear to be active.

I'm totally out of ideas, how I can fix this problem 😭

CoookieTM, 2025-06-11:
> I spent hours trying to get my generic ESP32-C3's working and finally from info on this forum, I hit the reset device and everything sparked into life.
> 
> [image]

I have the same esp as OP and hitting this console reset twice fixed my issue.

jwidess, 2025-06-22:
I was having the same issue with my [ESP32-C3 Super Mini](https://www.aliexpress.us/item/3256805675863451.html) board. I tried so many things, external power supply, resetting, flashing from different methods and directly with pytool, etc. The fix ended up being as simple as adding the following to my YAML config,
```yaml
wifi:
  output_power: 8.5dB
```
This seemed to have both fixed my improv serial issue alongside my `Event: Disconnected ssid='SSID' bssid=[redacted] reason='Auth Expired'` problem. 

Credits for this info:
- https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-1212591202
- https://community.home-assistant.io/t/unable-to-connect-to-wifi-auth-expired-and-association-expired/678570/2

AmritWafa, 2026-01-15:
With my ESP32-C3, I resolved this problem by changing the code to have:

1. platformio_options
2. manual_ip
3. output_power

```
esphome:
  name: stair-lights
  friendly_name: Stair Lights
  platformio_options:
    board_build.flash_mode: dio

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.6.x
    gateway: 192.168.1.x
    subnet: 255.255.255.0
  output_power: 8.5
```