esphome/esp-web-tools #466: Feature-Request: Serial console terminal

gitcnd, 2024-01-29:
Your type-and-send console is minimalistically handy, but there's no way to properly use my micropython device with it (e.g. so I can run "edit" to edit my files on the device and have the arrow keys work properly in the editor...)

How hard would it be to add the option to make it work like an actual terminal (e.g. like telnet)?

If I fixed this (added that feature), would my PR become accepted?

melkati, 2024-01-29:
It would be very useful for my project. +1

ldijkman, 2024-02-24:
Console clear button and clickable link detection

https://ldijkman.github.io/async-esp-fs-webserver/WebSerialMonitor.html

balloob, 2024-02-26:
@gitcnd it seems out of scope for ESP Web Tools. Terminal is provided to help people diagnose what is going wrong and pass simple commands. It is not meant as a full terminal.

closed 2024-02-26 by balloob