esphome/esp-web-tools #483: Getting "Invalid header" when trying to flash Seeed Studio XIAO ESP32C3 with custom firmware

mineshaftgap, 2024-03-08:
When I try and follow the [directions](https://esphome.github.io/esp-web-tools/#preparing-firmware) to add custom firmware, I get a infinite loop of:

```
invalid header: 0x434c4557
```

I used this command to make the merged firmware:

```
esptool.py --chip esp32c3 merge_bin \
  -o esp32c3.bin \
  --flash_mode dio \
  --flash_freq 40m \
  --flash_size 4MB \
  0x1000 .pio/build/seeed_xiao_esp32c3/bootloader.bin \
  0x8000 .pio/build/seeed_xiao_esp32c3/partitions.bin \
  0xe000 ~/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin \
  0x10000 .pio/build/seeed_xiao_esp32c3/firmware.bin
```

I noticed that when I build in Platformio/VSCode it showed:

```
{"offset": "0x0000", "path": ".../.pio/build/seeed_xiao_esp32c3/bootloader.bin"},
{"offset": "0x8000", "path": ".../.pio/build/seeed_xiao_esp32c3/partitions.bin"},
{"offset": "0xe000", "path": ".../.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin"}], "application_offset": "0x10000"}
```

So I also tried adjusting the above command to reflect a 0 offset for the bootloader like this:

```
esptool.py --chip esp32c3 merge_bin \
  -o esp32c3.bin \
  --flash_mode dio \
  --flash_freq 40m \
  --flash_size 4MB \
  0x0000 .pio/build/seeed_xiao_esp32c3/bootloader.bin \
  0x8000 .pio/build/seeed_xiao_esp32c3/partitions.bin \
  0xe000 ~/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin \
  0x10000 .pio/build/seeed_xiao_esp32c3/firmware.bin
```

Which produced this warning:

```
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash frequency setting. Use the --flash_frequency=keep option instead of --flash_frequency=40m in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
```

While the two different merged firmware byte counts were identical, I did run a diff and it does appear that they are different.

Thanks for any help anyone can give!

mineshaftgap, 2024-03-20:
@Jason2866 Thank you for your response. While esptool v4.8.dev3 did remove the warning, the same invalid header messages come up on boot no matter what I try. I have logged a ticket with [esptools](https://github.com/espressif/esptool/issues/968).

mineshaftgap, 2024-03-22:
@Jason2866 thanks for all you trying to help me!

In the end I finally discovered what was causing this and it was my user error, but also shed light on another bug that maybe should be addressed and I will open a new ticket for it. In my manifest I was referencing the path as "/fw/esp32c3.bin" where it should have been "fw/esp32c3.bin". Once I corrected the path everything worked fine.

I discovered this by trying to upload a hello world firmware on both my self-hosted ESP Web Tools and the official one and compared the web console output. As you can see below, mine is only writing 12 bytes vs the actual 305744 bytes. The bug (or feature missing as it may be) is that there appears to not be a test checking for the existence of the firmware referenced in the manifest or that the firmware size is greater than a certain byte count.

My output was:
```
install-dialog-BWZCBYvU.js:477 esptool.js
install-dialog-BWZCBYvU.js:477 Serial port WebSerial VendorID 0x303a ProductID 0x1001
install-dialog-BWZCBYvU.js:477 Connecting...
install-dialog-BWZCBYvU.js:477 .
install-dialog-BWZCBYvU.js:477 
install-dialog-BWZCBYvU.js:477 Detecting chip type... 
install-dialog-BWZCBYvU.js:477 ESP32-C3
install-dialog-BWZCBYvU.js:477 Chip is ESP32-C3 (revision 4)
install-dialog-BWZCBYvU.js:477 Features: Wi-Fi
install-dialog-BWZCBYvU.js:477 Crystal is 40MHz
install-dialog-BWZCBYvU.js:477 MAC: ec:da:3b:be:7a:d4
install-dialog-BWZCBYvU.js:477 Uploading stub...
install-dialog-BWZCBYvU.js:477 Running stub...
install-dialog-BWZCBYvU.js:477 Stub running...
install-dialog-BWZCBYvU.js:477 Manufacturer: 20
install-dialog-BWZCBYvU.js:477 Device: 4016
install-dialog-BWZCBYvU.js:477 Detected flash size: 4MB
install-dialog-BWZCBYvU.js:477 Erasing flash (this may take a while)...
install-dialog-BWZCBYvU.js:477 Chip erase completed successfully in 14.81s
install-dialog-BWZCBYvU.js:477 Not changing the image
install-dialog-BWZCBYvU.js:477 Compressed 12 bytes to 18...
install-dialog-BWZCBYvU.js:477 Writing at 0x0... (100%)
install-dialog-BWZCBYvU.js:477 Wrote 12 bytes (18 compressed) at 0x0 in 0.044 seconds.
install-dialog-BWZCBYvU.js:477 Leaving...
install-dialog-BWZCBYvU.js:807 HARD RESET
install-dialog-BWZCBYvU.js:807 DISCONNECT
install-dialog-BWZCBYvU.js:477 Initializing Improv Serial
```

The official one was:
```
esptool.js
create-esploader.ts:5 Serial port WebSerial VendorID 0x303a ProductID 0x1001
install-web-dialog.ts:189 Connecting...
esploader.js:203 .
esploader.js:203 
esploader.js:203 Detecting chip type... 
esploader.js:203 ESP32-C3
esploader.js:203 Chip is ESP32-C3 (revision 4)
esploader.js:203 Features: Wi-Fi
esploader.js:203 Crystal is 40MHz
esploader.js:203 MAC: ec:da:3b:be:7a:d4
esploader.js:203 Uploading stub...
esploader.js:203 Running stub...
esploader.js:203 Stub running...
esploader.js:203 Manufacturer: 20
esploader.js:203 Device: 4016
esploader.js:203 Detected flash size: 4MB
flash.ts:46 Erasing flash (this may take a while)...
esploader.js:203 Chip erase completed successfully in 14.703s
flash.ts:56 Not changing the image
flash.ts:56 Compressed 305744 bytes to 145205...
esploader.js:203 Writing at 0x0... (11%)
esploader.js:203 Writing at 0x13e4e... (22%)
esploader.js:203 Writing at 0x1e672... (33%)
esploader.js:203 Writing at 0x24566... (44%)
esploader.js:203 Writing at 0x2a66a... (55%)
esploader.js:203 Writing at 0x308a1... (66%)
esploader.js:203 Writing at 0x36593... (77%)
esploader.js:203 Writing at 0x3e4f2... (88%)
esploader.js:203 Writing at 0x4510a... (100%)
esploader.js:203 Wrote 305744 bytes (145205 compressed) at 0x0 in 2.465 seconds.
esploader.js:203 Leaving...
```

closed 2024-03-22 by mineshaftgap

Jason2866, 2024-03-22:
Great you found the reason!