esphome/esp-web-tools #487: Start Improv after succesfull installation

DustinWatts, 2024-03-19:
Hi people who take the time to read this 😄 !

I am a bit stuck/confused. Can any one tell me how do I start Improv after the installation is complete. And what needs to be in my code and what need to be on my website (where I host the web-tools for uploading the binaries.

Your answer or pointing me in the right direction is very much improved!

balloob, 2024-03-20:
You need to implement [Improv](https://www.improv-wifi.com/) in your firmware. If your firmware is slow on first boot, you can tweak the config. From the README:

> When a firmware is first installed on a device, it might need to do some time consuming tasks like initializing the file system. By default ESP Web Tools will wait 10 seconds to receive an Improv Serial response to indicate that the boot is completed. You can increase this timeout by setting the optional manifest key new_install_improv_wait_time to the number of seconds to wait. Set to 0 to disable Improv Serial detection.

ldijkman, 2024-04-22:
if you use arduino ide
in the library manager there is a library with improv example code

closed 2024-12-23 by balloob