esphome/esp-web-tools #488: Non-existent firmware will be upload, need to check it it exists before trying to flash

mineshaftgap, 2024-03-22:
In manifest.json if there is a typo, self-hosted ESP Web Tools will try and upload a non-existent firmware resulting in a boot loop of:

```
invalid header: 0x434c4557
```

More details can be found in a [previously related ticket](https://github.com/esphome/esp-web-tools/issues/483#issuecomment-2015479280). I suggest that checking for the existence of the file before trying to upload or in that is not possible to do a file size check.

balloob, 2024-04-01:
We check if the status code is correct:

https://github.com/esphome/esp-web-tools/blob/ec35b6aa693146432dad48b99b631b45bcae69b3/src/flash.ts#L119-L124

Web servers should return a 404 status code if the file is not found. If that is not the case, your web server is misconfigured.

closed 2024-04-01 by balloob

mineshaftgap, 2024-04-02:
Paulus, thanks for your response. My web server is not misconfigured and it does return 404s properly. I am not blocked by this issue, I just thought your team should know. Thanks.
[image]