esphome/esp-web-tools #493: "Visit Device" tries to use wrong IP when multiple networks are defined

Hiekkaharju, 2024-04-08:
Steps to reproduce :
-define web_server:
-define two or more networks(==SSID), at least one with manual_ip, (first) one with DHCP=without manual_ip
-start device when only the DHCP network is available (or it is clearly strongest)
-try to "visit device" - it will try to access the manual IP, not device_name.local (or dhcp IP) as it should

ESPHome web will in the same situation succesfully access e.g. Log, or install new fw, wirelessly automatically, apparently using the DHCP ip, - thus the bug is only about "visit device"

(see also https://github.com/esphome/esp-web-tools/issues/349 which is a different issue as this here happens even when the fixed ip has never been used)

balloob, 2024-04-09:
Visit device sends the user to the address as reported by the device. If your device doesn't report the correct name, it should be fixed in the firmware.

closed 2024-04-09 by balloob

Hiekkaharju, 2024-04-09:
Thanks for looking at this so rapidly.
But what exactly do you mean by "Reported by the device" ?
This is happening on a pure esphome - yaml - arduino esp32s3 device, no external components etc.
So from which esphome/arduino component "visit device" is getting the IP and how ?

wifi definition:
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
    - ssid:   ${pp200Ahname}
      password: !secret wifi_password
      manual_ip:
        static_ip: 192.168.4.10
        gateway: 192.168.4.10
        subnet: 255.255.255.0

and two of the sensors:
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
      id: ip
      internal: true
    ssid:
      name: ESP Connected SSID
      id: ssid
      internal: true

When the device is e.g. right after a wireless fw update still connected to !secret wifi_ssid -network, used to load the fw, "visit device" tries to go to 192.168.4.10, which won't work as the APs/routers in !secret wifi_ssid will give somehing starting 192.168.0.
If I manually go to actual dhcp IP, which I find out from e.g. the log output after the fw load or by requesting wirelessly the logs from the device, I see that those text sensors have the correct values, not 192.168.4.10 or ${pp200Ahname}.
And actually the 192.168.4..* / ${pp200Ahname} network does not even exist at that moment, as it is's an AP created by another esphome device only when it can't connect to to the same !secret wifi_ssid.
So why does "visit device" use some different method for determining the IP than e.g. getting the logs wirelessly?

balloob, 2024-04-09:
So that's a bug in ESPHome, not the web tools