esphome/esp-web-tools #497: Is it possible to flash esp-c6?

BasBaas08, 2024-04-10:
There is a closed issue mensioning there should be support for the ESP32-C6 SoC in the latest version. But when i try, it says the board is unsupported. is this a known issue?

jonasniesner, 2024-09-29:
Yes, you cam flash an esp32-C6

BasBaas08, 2024-09-29:
[image]

BasBaas08, 2024-10-16:
I have found that esp-web-tools uses esp-idf 4.4, and support for esp32c6 doesn't start until 5.0. So it's more of a question of when it will be updated to 5.0+

EmileSpecialProducts, 2024-12-08:
I do not see the Issue, I can program the ESP32-C6.
You can have a try at [this loaction](https://emilespecialproducts.github.io/Turtle-ESP-mouse-jiggler/firmware/upload.html)
and you can find the code at this [location](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler)
The only problem i have is that the C6 is not yet supported by platformIO :-(

balloob, 2024-12-23:
C6 is supported by ESP Web Tools but our example firmware doesn't support it.

closed 2024-12-23 by balloob