esphome/esp-web-tools #515: ESP32 S2 Mini / Adafruit ESPTool

TobiasPSP, 2024-07-10:
ESP32 S2 Mini uses dual UART USB and changes active USB port when switching to firmware upload mode. ESP Web Tool breaks with this board: 

Connecting via Web Serial works well, selection menu shows "ESP32-S2" (when in firmware upload mode), else "TinyUSB". 

"Prepare for First Use" and "INSTALL" both raise "Failed to initialize" after a few seconds of spinning wheel.

Browser-based "Adafruit ESPTool" (https://adafruit.github.io/Adafruit_WebSerial_ESPTool/) works like a charm. Apparently, this tool correctly handles the S2 Mini.

I'd like to either

a) understand how "Prepare For First Use" works and where the ESP Web Tools gets the four bin files that are mentioned in its manifest so I can use the Adafruit tool to provision S2 Mini with these files in the same way that "Prepare For First Use" would do if it worked correctly with S2 Mini.
b) better yet, get the ESP Web Tools to work with S2 Mini by adopting the technique used by Adafruit ESPTool

I cannot create ESPHome configuration and download its firmware, then upload with the Adafruit tool (which does works beautifully) since I need to provision 100s of boards. I need a generic reusable firmware that triggers "ADOPT" in ESPHome.

Any help or suggestion greatly appreciated. Many thanks.

EmileSpecialProducts, 2024-10-02:
This is linked to the [#375](https://github.com/esphome/esp-web-tools/issues/375) issue.
I have same problem that the 10.0.1 will not work on the WEMOS LOLIN S2.
But will work on the 8.0.6 version.
You can find my project [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler)
This also has a script for platformIO to create the Bin files.

spuder, 2024-11-03:
Thank you for your workaround. I've also confirmed that I'm unable to flash any Wemos D1 Mini S2 using version 9.x or 10.x of esp-web-tools. 

Downgrading to 8.0.6 is a good workaround.

HanYangZhao, 2024-12-06:
@spuder I'm having the same issue. How did you get the downgrade to work? I ran the server locally with version 8.0.6 but the web page keeps saying : 

The demo is not available because your browser does not support Web Serial. Open this page in Google Chrome or Microsoft Edge instead.

even if I'm using Chrome or edge

spuder, 2024-12-06:
You can see my working 0.8.6 code here. It is a Jekyll site

https://github.com/spuder/OpenSpool/blob/main/docs/index.md

EmileSpecialProducts, 2024-12-06:
>However with my Wemos Lolin S2 mini ESP32-S2 board on version 8.0.6 it simply says that your ESP32-S2 board is not supported

You can test the S2 board upload using this [link](https://emilespecialproducts.github.io/Turtle-ESP-mouse-jiggler/firmware/S2upload.html) 
And you can see what i have done at [this loaction](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler/tree/main/firmware)

EmileSpecialProducts, 2024-12-08:
[damz](https://github.com/damz) 
This [issue](https://github.com/espressif/esptool-js/issues/161#issuecomment-2524996331) is worked on by the ESPTool-JS team.
And is resoved, but a release still has to be made. 
Then the  ESP-Web-tool can be upgraded.

The1Percent, 2024-12-21:
Any idea when a release is coming? There are a lot of S2 minis waiting to be flashed at the moment.

EmileSpecialProducts, 2025-08-15:
It looks like it is resolved in the 10.1.1 release version. 
You can test it using this [link](https://emilespecialproducts.github.io/UltraWiFiDuck/upload.html)

sblantipodi, 2025-12-11:
10.1.1 still don't work here on Lolin ESP32-S2

[image]

is 10.1.1 supposed to contain the fix?