esphome/esp-web-tools #517: Allow log & Console to connect at different / configurable baudrate

marcelrv, 2024-08-22:
The web flasher has a 'Log & Console' option.
As not all applications use baudrate of 11500 for the serial communication, would be great if this could be configurable rather than fixed.

e.g. we have an application that needs to communicate only at 38400 due to compatibility with other systems.
Would be nice to be able to see the log based on that baudrate rather than that it appears that it is not working.

SalamunKawlam, 2024-12-29:
Agreed. We also have a bunch of using different baud rate for serial prints. So having an option to change the baud rate in the ui or (better yet having a baud rate parameter in the manifest file) would be a incredibly useful.

SalamunKawlam, 2024-12-29:
@balloob can you please take a look at this?

eried, 2025-03-07:
or maybe "autobaud" option :P

rainerfritz, 2026-02-25:
Seems to me same speed. But yes I missed the topic.