esphome/esp-web-tools #545: Chrome and Edge crash with esp-web-tools (including the demo)

yanshay, 2024-12-23:
Today I tried esp-web-tools after a few weeks of not touching it, and noticed both Chrome and Edge crash once the installation is complete.

It happens both with an application I'm writing as well with the example at https://esphome.github.io/esp-web-tools/ .

Just press Connect, select the port, it performs all the flashing and then the browser just crash and cleanly disappears.

Is it due to some new browser update that came out recently? something else?

yanshay, 2025-01-16:
I think I found what's causing this.

When I use esp-web-tools from the official published bundle, it crashes chrome right after flashing.
When I use esp-web-tools that I built locally the exact same html and binaries work well.

At least when the html and all other files are served locally.

thomaszlow, 2025-05-19:
@swoboda1337  I tried the esp-web-tools demo on Mac OS `15.4.1` and Chrome `136.0.7103.114` today and Chromes crashes like described by @yanshay immediately after flashing is finished. Is someone else facing the same issue?

In Chrome on Windows everything is working fine.

swoboda1337, 2025-05-19:
Does the console log show anything?

yanshay, 2025-05-19:
> Does the console log show anything?

If you refer to the crash, the browser just crash with all windows/tabs. So at least I don't know how to view the logs then.

thomaszlow, 2025-05-19:
This is really weird! Now, it's working.
Only an `install-dialog-CLL1RKXl.js?module:825 Improv initialization failed. Error: Improv Wi-Fi Serial not detected` error.

[image]

...

[image]

yanshay, 2025-05-19:
It's not weird - sometimes it's crashing and sometimes it doesn't. For me, most times it doesn't but from time to time it crashes.

BTW - I also have Mac, so it might be a Mac issue only.

I also don't know for sure, but I think when I hosted the code on my server it crashed much less than when loaded from packaged version from some public source I pulled it from.

yanshay, 2025-05-19:
As for Improv, I also had issues with Improv WiFi.
In my case I implemented the Improv protocol myself on the device so know exactly what it's doing.

With the Improv WiFi web site things worked perfectly, Improv WiFi post flashing didn't work. I noticed that the first bytes did not reach the device, maybe because they were sent immediately before the device could initialize the ports. I solved it by assuming the first command was sent and replied to Improv immediately, but I think the Improv code is not flexible enough on that front, should retry if does not get an response within some time, etc.

thomaszlow, 2025-05-20:
Hmm, although there is the error 

```console
install-dialog-CLL1RKXl.js?module:825 Improv initialization failed. Error: Improv Wi-Fi Serial not detected
    at install-dialog-CLL1RKXl.js?module:493:1210
```
in the console Improv WIFI is working for me. 
I get an error notification in the set up process in case I enter wrong WIFI credentials or a successful connection.

Somehow the crash situation is difficult to reconstruct on the Mac.

balloob, 2025-05-20:
I think that we sent the improv hello message too soon while the device is still rebooting. We should consider sending it a couple of times.