esphome/esp-web-tools #552: [Feature] Allow more than one firmware for each chip, with menu

gfwilliams, 2025-01-15:
esp-web-tools has been great at allowing users to easily flash the Open Source Espruino JS interpreter onto a device: https://www.espruino.com/Espressif+Flash

It's working great - the only problem is that multiple different devices can contain the same chip. For instance we have one generic firmware for the ESP32 dev boards, but then if a user has the ESP32 CYD (cheap yellow display) they would want a firmware that instantiates the display out the box.

You can't detect what the device is, I'd love to be able to do:

```
"builds": [
    {
      "chipFamily": "ESP32",
      "description": "Generic ESP32 Devboard",
      "parts": [ ... ]
    },
    {
      "chipFamily": "ESP32",
      "description": "ESP32 Cheap Yellow Display (CYD)",
      "parts": [ ... ]
    },
    {
      "chipFamily": "ESP32-C3",
      ...
```

and have esp-web-tools pop up a menu asking the user which firmware they'd like to install if more than one matched the chip...

Jason2866, 2025-01-16:
Why not providing the different firmware variants in the menu?

gfwilliams, 2025-01-17:
It feels like as we have the manifest and we list all different kinds of hardware in there, it would be a nice addition to allow different hardware with the same chips.

* It makes it a lot easier when making the website - we're not all having to duplicate adding the same chooser code and create `device_count * firmware_version` manifest.json files
* It's easier for users - when they're presented by a menu for which device to use it'll show only the devices that it *could* be. If you're connected to an ESP32S3 there's no point asking the user to choose from a list that's 90% populated with ESP32s

TD-er, 2025-01-17:
So what you're asking for is essentially a feature to "detect" the chip and trigger some update of the selector on the page.
A bit of a 'prequel' to what IMPROV does.

Possible issue I'm seeing is that you may need to turn the ESP into flashing mode to let the detection run which is also running to select the right part from the manifest.
However some ESP's like the ESP32-S2 when connected directly to USB (without USB to serial chip) will prompt a new USB device connect when putting the ESP into flash mode.

Would be a nice feature though, but I have absolutely no idea how much effort it would take to implement this extra hook.

gfwilliams, 2025-01-17:
> So what you're asking for is essentially a feature to "detect" the chip and trigger some update of the selector on the page.

Well, more to use an esp-web-tools provided menu, rather than updating something on the website...

Detecting the chip is done currently when you click `Install XYZ`, so I was proposing that at the point where you currently just select one firmware:

https://github.com/esphome/esp-web-tools/blob/main/src/flash.ts#L69

You instead check if more than one matches and show a menu to select which one to use: 

```
builds = manifest.builds.filter((b) => b.chipFamily === chipFamily);
if (build.length>1) 
  await builds = showBuildSelectorDialog(builds);

build = builds[0];
```

There's already the menu at startup (showing `Install XYZ, Logs & Console`):

[image]

I'm not sure how easy that is to reuse, but I guess I'd imagined something very similar could be done showing all the names of firmwares that match chipFamily in `manifest.builds`