esphome/esp-web-tools #583: android support

EmileSpecialProducts, 2025-04-07:
Can Android be supported.
See  esptool-js [Support Chrome on Android with web-serial-polyfill in the Example Pages](https://github.com/espressif/esptool-js/pull/144)

balloob, 2025-05-07:
You can load the polyfill before loading ESP Web Tools on your own website for Android support.

closed 2025-05-07 by balloob

balloob, 2025-05-10:
The way you append it, you are loading it after ESP Web Tools.

EmileSpecialProducts, 2025-05-10:
Is ia add it like this it sill does not work. 
Have you seen this work ?

```
<script src="https://cdn.jsdelivr.net/npm/web-serial-polyfill@latest/dist/polyfill.js"> </script> 
<script type="module" src="https://unpkg.com/esp-web-tools@10.1.0/dist/web/install-button.js?module"></script>

```