esphome/esp-web-tools #622: Installation failed on Chrome@MacOS

Mashiro69, 2025-08-10:
When I click the `Install esphome.esp_web_tools_example` Botton , It shows : `Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.`

```
◎ esptool --port /dev/cu.usbmodem21201 read-mac 
esptool v5.0.2
Connected to ESP32-C3 on /dev/cu.usbmodem21201:
Chip type:          ESP32-C3 (QFN32) (revision v0.3)
Features:           Wi-Fi, BT 5 (LE), Single Core, 160MHz
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                60:55:f9:7e:89:a4

Stub flasher running.

MAC:                60:55:f9:7e:89:a4

Hard resetting via RTS pin...
```

```
esptool --port /dev/cu.usbmodem21201 erase-flash
esptool v5.0.2
Connected to ESP32-C3 on /dev/cu.usbmodem21201:
Chip type:          ESP32-C3 (QFN32) (revision v0.3)
Features:           Wi-Fi, BT 5 (LE), Single Core, 160MHz
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                60:55:f9:7e:89:a4

Stub flasher running.

Flash memory erased successfully in 0.0 seconds.

Hard resetting via RTS pin...
```

Chrome version: 139.0.7258.67 (arm64)
MacOS version: macOS Tahoe Beta 26.0 (25A5316i)

screekworkshop, 2025-08-11:
https://espressif.github.io/esptool-js/ break too
[image]

Mashiro69, 2025-08-11:
Now I use `esptool write-flash` to instead of its

screekworkshop, 2025-08-11:
this maybe work:
```
open -a "Google Chrome" --args --disable-features=SerialSplitDtrAndRts
```

https://github.com/espressif/esptool-js/issues/206
https://issues.chromium.org/issues/420689824#comment11

screekworkshop, 2025-08-12:
[image]
[image]

https://chromium.googlesource.com/chromium/src/+log/139.0.7258.67..139.0.7258.128?pretty=fuller&n=10000

It has been fixed in the latest version released on 139.0.7258.128. This behavior is now disabled by default, so no additional parameters are required. Just click “Update” in “About.”

Currently, the traces for this issue are as follows: Chrome version 139.0.7258.67 has a bug that requires manual parameter settings to work, while subsequent and previous versions are normal.