esphome/esp-web-tools #624: SPIFFS not flashed when using merged binary with ESP Web Tools

DanzoCZ, 2025-08-13:
When creating a merged_firmware.bin from bootloader.bin, partitions.bin, boot_app0.bin, app.bin, and spiffs.bin and flashing it through ESP Web Tools, the SPIFFS partition is not correctly flashed or mounted.

Steps to reproduce:

Build firmware in Arduino IDE with "No OTA (2MB APP / 2MB SPIFFS)" partition scheme.

Generate spiffs.bin with mkspiffs.

Merge binaries:
`esptool --chip esp32 merge-bin -o merged_firmware.bin \
  --flash-mode dio --flash-freq 80m --flash-size 4MB \
  0x1000 app.bootloader.bin \
  0x8000 app.partitions.bin \
  0xe000 boot_app0.bin \
  0x10000 app.bin \
  0x210000 spiffs.bin
` 
Flash merged_firmware.bin via ESP Web Tools.
SPIFFS mount fails with: 
`E (1455) SPIFFS: mount failed, -10025
`

Expected behavior:
SPIFFS should be flashed correctly and mount without errors.

Actual behavior:

SPIFFS partition remains empty or corrupted.
Merging works for app.bin but not for spiffs.bin.
Flashing via esptool.py with separate offsets works as expected.

Additional info:

Board: ESP-WROVER-B (4MB flash)
ESP32 Arduino Core: 2.0.11
ESP Web Tools: latest
spiffs.bin size matches partition table.
Issue occurs regardless of partition scheme.

EmileSpecialProducts, 2025-08-15:
Have you managed to program the Bin file using the [esptool-js](https://espressif.github.io/esptool-js/).
This will give an indication if it is the ESP Web Tools or the merged_firmware.bin file.

EmileSpecialProducts, 2025-08-16:
The Arduino IDE is great to start programming, but you do not have the control that ,you have in the PlatformIO IDE.
In this case I have a created a post processing script that will create a littlefs.bin file and the files that are needed for the ESP-Web-tool programming