esphome/esp-web-tools #629: Flash memory always overwritten on install

joetemus, 2025-09-03:
Trying to set things up so that it will not overwrite the flash memory on install. I have the erase device button enabled and deselected, and it still overwrites the flash memory. Any thing I can do to prevent that? using merged bin files with no_reset selected.

joetemus, 2025-09-03:
Trying to make it so I have a flash value I pull to know what software version that person has paid for. If it erases that location then I have no way of knowing if they have the correct value when they update the firmware to get new features.

RowdyRutabaga, 2026-01-18:
I have the same issue; erasing flash clears all of my customer's configuration data.  I've tried the latest version and still observe the issue.  I'm going to try to figure out what is causing this.  If anyone knows what version they last observed the don't erase flash feature working, please leave a comment.